catgiggle/OctoPrint-GpioControl #1: Plugin fails to load if another GPIO plugin is enabled

alefchak, 2021-02-17:
I had another GPIO plugin installed (OctoLight) and wanted to try this one out. It failed to load completely with the following stack trace:

```
2021-02-16 19:40:20,546 - octoprint.plugin.core - ERROR - There was an error loading plugin gpiocontrol
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1526, in load_plugin
    plugin.load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiocontrol/__init__.py", line 150, in __plugin_load__
    __plugin_implementation__ = GpioControlPlugin()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiocontrol/__init__.py", line 19, in __init__
    GPIO.setmode(GPIO.BCM)
ValueError: A different mode has already been set!
```

Disabling the other plugin allowed this to load, but maybe an error on the plugin settings page or something would be better.

catgiggle, 2021-02-17:
Thanks for issue. I will fix this as soon as possible.

catgiggle, 2021-02-17:
Compatibility with other plugins that uses RPi.GPIO library added.

closed 2021-02-17 by catgiggle