catgiggle/OctoPrint-GpioControl #3: Request - Navbar Buttons

Pheelix, 2021-02-17:
Any chance of adding the GPIO Controls to the navbar?  I know there are some already out there for navbar, but they only control one gpio or type of thing, like psu or lights and requires installing 2 different plugins.  I have been waiting for someone to make a simple add X many gpio plugin.

Also it would be neet if you could make the icons change color or look when on or off.  Like when you turn the power or light off, it would turn to a dark icon.  And when on it would turn to a light or colored icons.

catgiggle, 2021-02-17:
> Any chance of adding the GPIO Controls to the navbar?

Yes. I thought about adding switch/select in settings named "Visibility" with two options: "sidebar" and "navbar". This plugin allows to adding as many buttons as you want. But navbar has not enough space for all of them.  Thank to this it is possible to manage the space of navbar. But it is not enough, because viewport can change at any time (eg. mobile). So if the buttons don't fit into the navbar, they should appear in the sidebar, even if they are configured to show in navbar (they has to be visible somewhere).

> like psu or lights and requires installing 2 different plugins

Please note that PSU Control is a bit more advanced plugin than just switching on/off. Author of this plugin did a great job and I think there is no reason to duplicate its functionality. But of course if you just need to on/off your printer, then I will be glad if you use GPIO Control.

> Also it would be neet if you could make the icons change color or look when on or off. Like when you turn the power or light off, it would turn to a dark icon. And when on it would turn to a light or colored icons.

Please note that not all icons have their dark counterparts, so they could be colored. But it means two additional colorpickers in settings, right?

BTW, thanks for your ideas! Honestly, I thought about it myself too, but I wanted this plugin to be as simple as possible. Therefore I will implement these ideas for sure, but probably as another, little more complex, plugin.

Pheelix, 2021-02-18:
> BTW, thanks for your ideas! Honestly, I thought about it myself too, but I wanted this plugin to be as simple as possible. Therefore I will implement these ideas for sure, but probably as another, little more complex, plugin.

Yup np, just throwing out some upgrade ideas.  And yes PSU Control is a very nice plugin.  As I like the simple gpio control you made, I will probably keep using PSU Control for its advance functions.  But I do like your plugin for being able to control all my other gpio stuff.

Looking forward to seeing a advanced version of GPIO Controls if you are planning something like that.

closed 2021-02-20 by catgiggle