catgiggle/OctoPrint-GpioControl #5: GPIO inputs

doe57716, 2021-02-20:
Is it possible to add inputs for things like physical buttons for this? The enclosure plug-in already does this but I find it’s interface to be very clunky, and this plugin is a lot more polished to me.

catgiggle, 2021-02-20:
Please provide more details how it should looks and works. Which plugin do you mean?

doe57716, 2021-02-20:
There is this plugin:
https://plugins.octoprint.org/plugins/enclosure/, which has options for
gpio inputs and outputs, such as attaching a relay to gpio and a button and
having the button control the relay, but I find that it’s interface is very
clunky.

catgiggle, 2021-02-20:
Very nice plugin! In GPIO Control you are also able to connect relay and control them. I'm not sure what is purpose of controlling relay via physical button with GPIO Control in the middle. Just connect button directly to relay. You can also connect GPIO in parallel. Thanks to this you will be able to control relay via physical button and GPIO Control too.

doe57716, 2021-02-20:
Thanks for the tip!

closed 2021-02-20 by catgiggle