catgiggle/OctoPrint-GpioControl #6: Off Button is almost not visible

DarthVad3r013, 2021-02-20:
Hello,
The plugin works fine but the off button is almost not visible on the browser.
I try with chrome on pc and mobile. The issue is the same.
I don't try with firefox or edge. 

[image]

[image]

catgiggle, 2021-02-20:
Thanks for issue. This is interesting because I tested this on Firefox and Chrome. On both everything looks fine. But I will investigate what is going on.

catgiggle, 2021-02-20:
Got it! Look for a version 1.0.2 which fix this

DarthVad3r013, 2021-02-20:
It seems working with the 1.0.3 
Thanks a lot

closed 2021-02-20 by DarthVad3r013

JeffreyFalgout, 2021-02-20:
Works for me too, thanks!

commit d757d3b7 (2021-02-20): Fix #6: Off Button is almost not visible
commit cc9376b8 (2021-02-20): Fix #6: Off Button is almost not visible