catgiggle/OctoPrint-GpioControl #10: Configuration issues.

simonkitrick, 2021-02-21:
I am running Octopi 0.18.0 and GPIO Control 1.0.7.

When I add pin assignments on the configuration they are shown on the main Octoprint screen with ON and OFF buttons greyed out.

When I go into the configuration screen again and delete the original pin assignment (and save) the original assignment is still on the main Octoprint screen.

I have tried restarting Octoprint and rebooting the system. The problem remains.

Am I dong something wrong?

All the best,

Simon.

catgiggle, 2021-02-21:
It's little strange that you was able to save settings once, but at the second one there was some problem. I will be grateful for more details. Especially logs from your instance of OctoPrint. Probably there is problem with Python compatibility.

simonkitrick, 2021-02-22:
Hi,

I'm new to github. What does 'assigned' indicate?

Thans.

catgiggle, 2021-02-22:
It's just to mark that currently the issue needs you attention. As soon as you provide more details (see my comment above) you can assignm issue to me. If you don't, I will probably do it myself :)

simonkitrick, 2021-02-22:
Sorry, 

I sent a copy of the log directly to your email address.

Here is a copy.

[octoprint.log](https://github.com/catgiggle/OctoPrint-GpioControl/files/6021455/octoprint.log)

Simon.

simonkitrick, 2021-02-22:
Hi,

I cannot see a checkbox to assign the task to you.

Simon.

catgiggle, 2021-02-22:
Thanks a lot! Now I see where the problem is.

catgiggle, 2021-02-23:
Fixed with v1.0.8.

closed 2021-02-23 by catgiggle