catgiggle/OctoPrint-GpioControl #9: [Bug] Update loop with v1.0.5

fnsign, 2021-02-21:
Hi,

this mornig Octoprint notified me of an update to v1.0.5. It confirm the update and after the installation process the notification reappears and the previous installation (1.0.4) is valid.

catgiggle, 2021-02-21:
Fixed. I forget to change version in code in previous release. Please let me know if it helps.

fnsign, 2021-02-21:
Hi, thanks for the quick response and providing the 1.0.6. But it still doens't work. After conducting the update, after restart it again requests to update.

catgiggle, 2021-02-21:
You are right. It should goes to master. Ugh I had to create next release and next version. I wanted to add a prerelease channel to avoid a ton of new versions on the stable channel but this change itself is a ton of new versions. Hope 1.0.7 will be the last one.

fnsign, 2021-02-21:
Now the update works!
Thanks again for your quick response and continue that great work!

closed 2021-02-21 by fnsign