catgiggle/OctoPrint-GpioControl #18: Loses settings on shutdown

Olie426, 2021-03-07:
I set up the Pi4 to control a 2 relay module. Worked fine during testing. Shut down the system to put the case back together and rebooted. Now the side menu shows the 2 switches but they are greyed out, no control. Only way I have found to bring tyhem back is to clean up data for plugin and then create the switches again. Once rebooted, backed to greyed out and no control. Any suggestions?

catgiggle, 2021-03-08:
Probably the settings was stored correctly, since everything worked fine before the reboot. Please send me your `octoprint.log` (Settings > OctoPrint > Logging). This will definitely help me track down the problem.

Olie426, 2021-03-08:
I tried to email the log but I guess it's too big. Hopefully, attaching it here works. Thank you.
[octoprint.log](https://github.com/catgiggle/OctoPrint-GpioControl/files/6101381/octoprint.log)

catgiggle, 2021-03-10:
Thanks for your log. It was very useful. I analysed them it founds out that the problem is related with below bug.
```
2021-03-07 15:58:28,829 - octoprint.server.api.settings - ERROR - Could not save settings for plugin GPIO Control (1.0.8)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 1034, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiocontrol/__init__.py", line 65, in on_settings_save
    GPIO.cleanup(pin)
ValueError: The channel sent is invalid on a Raspberry Pi
2021-03-07 15:58:30,023 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin gpiocontrol
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 98, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_gpiocontrol/__init__.py", line 167, in on_api_get
    states.append("on" if GPIO.input(pin) else "off")
RuntimeError: You must setup() the GPIO channel first
```
I will fix this bug with next version.

BrotherMG, 2021-03-11:
I also had the problem, when I saved the settings and refreshed the setttings where gone.
It was because of special characters. I'm from Austria and in German langage we are using ä, ö ,ü (just as example).
Hope this helps.