catgiggle/OctoPrint-GpioControl #23: Not able to turn off GPIO

Ender-08, 2021-03-13:
I have no clue what to do because I am able to turn of the GPIO pin but not turn it off.
[image]

ertraid, 2021-03-14:
Same here.. Octoprint 1.5.3 on docker (official image.)

catgiggle, 2021-03-15:
What happens after turning on?

Ender-08, 2021-03-15:
The gpio turns on but you can turn it off because the off button isn't active.

catgiggle, 2021-03-16:
Could you send me your `octoprint.log` (Settings > OctoPrint > Logging). This will definitely help me track down the problem.

ertraid, 2021-03-17:
[octoprint.log](https://github.com/catgiggle/OctoPrint-GpioControl/files/6153114/octoprint.log)

Here you go.. tried to turn on/off leds just before taking the log, so should be reflected. As far as I can see the problem is that your toggle switch does not toggle... the "ON" option remains active even after pressing it, so you cannot turn off.. I turned off the leds via a an alternative mqtt method and then can turn them on again via your plugin... but always on, never off..
Hope it helps.

catgiggle, 2021-03-17:
Thank you for the `octoprint.log`! Unfortunately there is nothing suspicious. I've checked whole file and still I don't know where is the problem. Have you tried with another pin, eg. GPIO3? If it doesn't work I will be very grateful for checking another plugin (eg. OctoLight, PSU Control). Maybe the problem is more general and not related only with GPIO Control.

plugowski, 2021-05-16:
Bump. Same issue. When click on ON button those requests goes to apI:

POST {id: 0, command: "turnGpioOn"}
GET ["off"]

Seems like GPIO status is not updated, but it works - I use it to turn on LEDS on hotend. Unfortunatelly I can't turn it off from octoprint, I can login by ssh to raspberry and manually turn off GPIO.

plugowski, 2021-05-17:
I've also checked GPIO state via SSH, so checking GPIO.input for specific pin always return 0.

```
>>> GPIO.output(18, 0)
>>> GPIO.input(18)
0
>>> GPIO.output(18, 1)
>>> GPIO.input(18)
0
```

And what is funny. It's only occurs for pin 18 (my RPi 2 model B). I tested rest pins and they return proper values...

```
#1 (on): 1
#1 (off): 0
#2 (on): 1
#2 (off): 0
#3 (on): 1
#3 (off): 0
#4 (on): 1
#4 (off): 0
#5 (on): 1
#5 (off): 0
#6 (on): 1
#6 (off): 0
#7 (on): 1
#7 (off): 0
#8 (on): 1
#8 (off): 0
#9 (on): 1
#9 (off): 0
#10 (on): 1
#10 (off): 0
#11 (on): 1
#11 (off): 0
#12 (on): 1
#12 (off): 0
#13 (on): 1
#13 (off): 0
#14 (on): 1
#14 (off): 0
#15 (on): 1
#15 (off): 0
#16 (on): 1
#16 (off): 0
#17 (on): 1
#17 (off): 0
#18 (on): 0
#18 (off): 0
#19 (on): 1
#19 (off): 0
#20 (on): 1
#20 (off): 0
#21 (on): 1
#21 (off): 0
#22 (on): 1
#22 (off): 0
#23 (on): 1
#23 (off): 0
#24 (on): 1
#24 (off): 0
#25 (on): 1
#25 (off): 0
#26 (on): 1
#26 (off): 0
```

plugowski, 2021-05-18:
@YeetTheLettuce @catgiggle @ertraid  I fixed issue with my latest PR: https://github.com/catgiggle/OctoPrint-GpioControl/pull/34